mcci-catena/Catena-Arduino-Platform #24: Documentation and examples in wrong place

terrillmoore, 2018-02-18:
The documentation and examples are in the wrong place, for some reason. They're partially at https://github.com/mcci-catena/Catena-Sketches, partially at https://github.com/mcci-catena/Catena-Sketches/tree/master/catena4450m101_sensor.  We should move the example sketches here (at least the relevant ones), and add an Adafruit-like tutorial on getting started with the hardware.

RussNelson, 2018-02-19:
I started to write a step-by-step, and found myself referencing just two documents: the hardware assembly PDF and catena4450m101_sensor/README.

terrillmoore, 2018-03-26:
This definitely needs to be address for V1.0.0.

vicatcu, 2018-06-16:
@RussNelson consider editing your reply so that the "Catena4450m101 README" linke points to https://github.com/mcci-catena/Catena-Sketches/blob/master/catena4450m101_sensor/README.md (it's a broken link currently)