mcci-catena/Catena-Arduino-Platform #27: Catena4551m101~104 is set up as Catena4550 during provisioning

sjpark-mcci, 2018-04-27:
After giving a command "system configure platformguid xxx...xxx" during provisioning Catena4551m101~104, the platformguid is always switched to Catena4550 platformguid. This is because in "src/lib/stm32/catena455x/Catena455x_GetPlatformForID.cpp", m101 ~104 is not added in CATENA_PLATFORM(); chooses the default which is Catena4550.

const CATENA_PLATFORM (* const Catena455x::vPlatforms[]) =
	{
	// entry 0 is the default
	&gkPlatformCatena4550,
	&gkPlatformCatena4551,
	};

To fix this, m101~104 need to be added in the above code and also in the header file. Similar bugs were found with samd code. The following files need to be fixed.

src/Catena_Platforms.h
- need to add Catena4551m101~104
src/lib/samd/catena4450/Catena4450_GetPlatformForID.cpp
- need to add Catena4450m103~104
src/lib/samd/catena4460/Catena4460_GetPlatformForID.cpp
- need to add Catena4460m103~104
src/lib/stm32/catena455x/Catena455x_GetPlatformForID.cpp
- need to add Catena4551m101~104

terrillmoore, 2018-04-27:
Is there anything for me to review? Have we created a pull request?

terrillmoore, 2018-04-28:
Changes are made on my side, doing test compiles.

commit 1f79e529 (2018-04-28): Merge pull request #31 from mcci-catena/issue27

Fix #27: add defined platforms
commit ec72dce2 (2018-04-28): Fix #27: add defined platforms
closed 2018-04-28 by terrillmoore