mcci-catena/Catena-Arduino-Platform #100: CatenaStm32L0 is misnamed

terrillmoore, 2018-11-24:
This class name implies that this is for any STM32L0-based design.

But in fact, this is specifically for Murata-L082-based feather-like modules. As we will support other modules in the future that may contain modules from other vendors, we should call this something like `CatenaMurata078` (or use `CatenaCMWX1ZZABZ_078`, which is the official name).

terrillmoore, 2018-11-27:
@chwon64 Any preferences on the name?  BTW, perhaps we should think about how we'd handle the AcSIP modules. There are two: S76S (LoRa only) and S76G (LoRa plus GPS). They use STM32L07x CPUs (not L082).

terrillmoore, 2018-11-27:
Yes, but CatenaStm32L0 it presently has the pin-definitions for the Murata module, and some assumptions about how MCCI uses the Murata module on a board. So maybe #101 was correct (because it is a fact that `CatenaStm32L0` really has a bunch of Murata-specific info), but the right response would have been to move the Murata-specific info to that layer and rename `CatenaStm32L0LoRa` to `CatenaMurata078`?  Then future AcSIP (or other) STM32L0 modules could inherit from CatenaStmL0, have another abstract class that represents the module and the wiring as seen from the module, and then have concrete classes to represent the boards?

terrillmoore, 2018-11-27:
Here's my mental reference model -- in practice there may be extra layers, but it simplifies to this. Suggests that the class hierarchy should be similar.

[image]

chwon64, 2018-11-27:
Okay, I will rename `CatenaStm32L0LoRa` to `CatenaMurata078` class and move out Murata specific to `CatenaMurata078`.

chwon64, 2018-11-27:
Accidentally closed.

chwon64, 2018-11-28:
Moved all pin definition from `CatenaStm32L0' class to Catena boards specific classes.

closed 2018-11-28 by chwon64