mcci-catena/Catena-Arduino-Platform #103: Refactor STM32 classes to be more like Samd21

terrillmoore, 2018-11-24:
Samd21 classes use an elaborate technique to avoid conditional compiles, based on the C compiler's ability to optimize out globals that aren't used.

1. The table of platforms contains all the known platforms.
2. The actual class `Catena4450`, etc., has a `getPlatformTable()` method that returns the platforms supported by the device.
3. Startup code calls that as needed.
4. Other platforms in the vector aren't referenced, so they're discarded.

STM32 classes instead have a table that's constructed with `#if`. This is much more fragile.

Please refactor the Stm32 classes to use the Samd21 technique.

chwon64, 2018-11-27:
Do you also want to remove `#if` when declare table instances as below?

https://github.com/mcci-catena/Catena-Arduino-Platform/blob/4ccdeff9d6481a8cbb6efd9eabc3cf814b8d7c50/src/lib/stm32/CatenaStm32_gk_PlatformHierarchy.cpp#L71-L87

If we remove this `#if`, these table instances are constructed and actual code size will be increased.

terrillmoore, 2018-11-27:
The linker is supposed to remove the unused `CATENA_PLATFORM` instances, because there will be no reference to them (this is the `--gc-sections` option).  You can compare size after you take out the `#if`s; size shouldn't change.  That's why I used this sort of strange format rather than a vector.

chwon64, 2018-11-27:
I checked the linker doesn't remove the unused `CATENA_PLATORM` instances. I also checked `--gc-sections` option applied when build STM32.

chwon64, 2018-11-28:
All changes are merged.

closed 2018-11-28 by chwon64