mcci-catena/Catena-Arduino-Platform #119: Catena 4612 and 4801 - LoRaWAN begin and provisioning failed

dhineshkumarmcci, 2019-01-04:
We have tested the latest “Catena-Arduino-Platform” with Catena devices.
The latest library works well with all devices except 4612 and 4801.

For Catena 4612 and 4801, we find below issues:
i.	LoRaWAN begin failed
ii.	Feeding the provisioning commands failed

Please find the test results below:

**Log for Catena 4612**
```
Catena 4612 test01 V0.1.0
SYSCLK: 32 MHz
USB enabled
LoRaWAN init: ?CatenaStm32L0::LoRaWAN::begin: Arduino_LoRaWAN:begin failed
?Catena461x::LoRaWAN::begin: Super::begin() failed
failed
CPU Unique ID: 08-20-47-04-32-37-47-04-33-34-37-39-47-00-43-00
EUI64: 00-02-cc-01-00-00-01-97
Platform Flags: 0x00007019
Operating Flags: 0x00000001
FLASH found, put power down
No mods detected
vBat: 2030 mV
vBus: 1066 mV
BME280: T: 29 P: 101662 RH: 37
Si1133: 494 IR, 165 White, 3 UV
EV_JOINING
EV_TXSTART
OK
EV_TXSTART
00-02-cc-01-00-00-01-97
OK
EV_TXSTART
?-1
EV_JOINED
NwkID: 00000013 DevAddr: 26012d3f
EV_TXSTART
```

**Log for Catena4801**
```
Catena 4801 Modbus Simple Host V0.1.0
LoRaWAN init: ?CatenaBase::GetProvisioningStyle: failing
?CatenaStm32L0::LoRaWAN::begin: Arduino_LoRaWAN:begin failed
?Catena480x::LoRaWAN::begin: Super::begin() failed
failed
message cycle time 30 seconds for 30 messages
CPU Unique ID: 08-20-47-04-33-37-47-04-35-35-31-31-29-00-1c-00
EUI64: 00-02-cc-01-00-00-02-3b
Platform Flags: 0x00013009
Operating Flags: 0x00000001
***No mods detected***
vBat: 1901 mV
Please wait while searching for MODBUS SLAVE
ERROR: MODBUS SLAVE UNAVAILABLE
Buffer size: 5
EV_JOINING
?CatenaBase::GetOtaaProvisioningInfo: failing
send buffer failed
not provisioned, idling
?-1
?-1
```

For diagnosis purpose, we have cloned the “Catena-Arduino-Platform” from branch “catena461x_adc”.
The result was good and as expected. The LoRaWAN provisioning worked.

terrillmoore, 2019-01-04:
Which devices did you test (not just failures)?

 

--Terry

dhineshkumarmcci, 2019-01-04:
We have tested with Catena 4450, 4460, 4551, 4612 and 4801. The failure is seen only with 4612 and 4801.

commit 04ed234f (2019-01-16): Merge pull request #120 from mcci-catena/issue119

Need to set up LoRaWan command before call Arduino_LoRaWAN::begin(). Fix #119.
closed 2019-01-16 by terrillmoore