mcci-catena/Catena-Arduino-Platform #125: ReadAnalog failed on selecting lower System Clocks

dhineshkumarmcci, 2019-02-13:
We tested Catena boards 4612 and 4801 at lower System Clocks (no USB) **2.097 MHz** and **4.194 MHz**. We see that the ReadAnalog failed with these two system clocks.

Further, we see function call `AdcCalibrate` returns `false` under the function `CatenaStm32L0_ReadAnalog` in **CatenaStm32L0_ReadAnalog.cpp**.

terrillmoore, 2019-02-13:
@dhineshkumarmcci Please check the data sheet, and also look at the logic - are we timing out because of slow clock, or otherwise malfunctioning? Is the ADC not returning an value, returning an error bit, or some other symptom? Quote code in the issue (learn how to do that, I do that a lot) so that the reviewer doesn't have to dig through code. Thanks.

dhineshkumarmcci, 2019-02-13:
@terrillmoore Thank you for your suggestion and feedback. I will be checking out the datasheet and will debug the issue further. I have quoted the code (that returns `false`) in this issue.

https://github.com/mcci-catena/Catena-Arduino-Platform/blob/259605b684cbbd797eb6e7e671c6e52efa49a448/src/lib/stm32/stm32l0/CatenaStm32L0_ReadAnalog.cpp#L114

https://github.com/mcci-catena/Catena-Arduino-Platform/blob/259605b684cbbd797eb6e7e671c6e52efa49a448/src/lib/stm32/stm32l0/CatenaStm32L0_ReadAnalog.cpp#L209-L230

terrillmoore, 2019-02-13:
@dhineshkumarmcci Let's think about this. What does line 223 do?  It's a timeout, right? 

Would it not make sense to increase the timeout at line 223 and see if it perhaps simply takes longer when running at a slower clock rate? 

That way you can give us more information about the problem. Or you could eliminate the timeout and simply print out (or save) the actual time it takes... if it hangs you know it's really broken...

terrillmoore, 2019-02-15:
I encounter the same thing on the 4612 -- changing the timeout did not seem to help, I still get:

```console
?CatenaStm32L0::ReadAnalog(2): CatenaStm32L0_ReadAnalog() failed
vBus:    4094 mV
BME280:  T: 26 P: 97398 RH: 13
Si1133:  390 IR, 136 White, 6 UV
?CatenaStm32L0::ReadAnalog(3): CatenaStm32L0_ReadAnalog() failed
vBat:    4094 mV
```

(This is using the new catena4612_simple.ino sketch).

terrillmoore, 2019-02-15:
Oh, I see, it's ReadAnalog() -- many of these functions have timeouts and probably *all* are incorrect.  I will prepare a patch that parameterizes all the timeouts.

terrillmoore, 2019-02-17:
I patched all to use a defined constant -- and even with a delay of 1000 milliseconds was unable to get calibration to complete. The STM32L0 reference RM0376 on page 298/299 says:

> Refer to Section 7: Reset and clock control (RCC) on page 170 to see how PCLK and ADC asynchronous
clock are enabled

But there is no mention of the "ADC asynchronous clock" elsewhere in the manual. This seems to be an error.  Switching to use PCLK by setting `ADC_CFGR2_CKMODE_0` (binary 01) also does not help.

The only thing that helps is running with a clock of 16 MHz or greater.

commit f4c51075 (2019-02-19): Merge pull request #131 from mcci-catena/issue125

Fix #125: Turn on and off HSI clock if system clock use MSI clock.
closed 2019-02-19 by terrillmoore