mcci-catena/Catena-Arduino-Platform #132: Need to improve STM32 power consumption

dhineshkumarmcci, 2019-03-10:
Sleep power consumption of Catena 4801 is 54 uA and for Catena 4612 is 34 uA with the latest version of BSP and library. The power consumption has to be reduced further.

dhineshkumarmcci, 2019-03-21:
We have worked on the power consumption during sleep. We tried resetting the GPIO pins before sleep and able to achieve average power consumption of **6 uA** for Catena 4801 and **24 uA** for Catena 4612. We made further research and see that **PB13** is the GPIO pin that reflect in Power consumption. 

We tried resetting only PB13 and the result is almost similar to above result. I have committed the changes to the branch '_issue132_'. 

I also like to add up that PB13 is configured as SPI2 SCLK in STM32 devices. Instead of resetting in library, this pin can be handled as digital pin in the top level sketch. In this case the average power is similar to the above result. But the maximum power is upto 36 uA, whereas in other cases it is less than 30 uA.

I have tried out the following test cases:

1. No GPIO pin has been reset
2. Reset all GPIO pins
3. Reset only GPIO port B 
4. Reset the GPIO pin PB13
5. Handle PB13 as digital pin in the sketch

The average power consumption of Catena 4612 for the case 2, 3, 4 and 5 is **24.4 uA, 24.1 uA, 25 uA** and **25.4 uA** respectively. I see a better result with **case 3** in all the sleep cycles.

svelmurugan92, 2019-04-05:
If, We use Standby as sleep mode. 
Wake-up event from Standby sleep mode is similar to reset.

> After waking up from Standby mode, program execution restarts in the same way as after a
Reset (boot pins sampling, vector reset is fetched, etc.).

So, Every-time exiting from Standby mode cause reinitialize the program.
In Catena 461x/4801 you can see, the log are repeating. 
Also, boot count is incremental in console. 

Looks, We might need to use StopWithLowPowerRegulator as sleep mode. 

Please share your thoughts.

terrillmoore, 2019-04-05:
It really depends on how much work you want to do in the application. If you can save state in FRAM, you can power down completely, and just restore state. However, we don't support this yet with the LMIC. My recommendation is that you use STOP mode for sleeping for now (and set the PWR_CR.ULP and PWR_CR.LPSDSR bits -- bits 9 and bit 0).  If that's what you mean by StopWithLowPowerRegulator, then that's correct. That's not an STM32L0 datasheet term, so I'm referring directly to the terms used in section 6.3.9.  It apppears to me that we need to set bits 9 and 0 in PWR_CR before sleeping.

chwon64, 2019-04-06:
It looks we need to revert changes in HAL_PWR_EnterSTANDBYMode() because changes are not working.

dhineshkumarmcci, 2019-04-08:
According to datasheet, it is said that the behavior of Standby mode is to reinitialize the program. Due to this reason, we configured the sleep mode to **StopWithLowPowerRegulator**. 

Also added the required changes in Arduino_Core_STM32 **stm32l0xx_hal_pwr.c** and created a issue [#76](https://github.com/mcci-catena/Arduino_Core_STM32/issues/76) in Arduino_Core_STM32.

commit 0cef7ce0 (2019-04-08): Fix #132: Configure Sleep mode to 'StopWithLowPowerRegulator'.