mcci-catena/Catena-Arduino-Platform #172: Catena 4460 questions

trlafleur, 2019-06-04:
1) I'm using a 4460, but need to add an I2C device at addr: 0x50 where the Fram is located...
How can I reassign the Fram address in SW to a new location with a 24LC64 device? I can re address part on the board, but what do I need to change to select the FRAM64 device in the code base?

2) I noticed on start up, I see LoRaWAN Init: message, where is the ?CatenaBase::GetProvisioningStyle: infor comming from.
~~~
Catena 4460 sensor1 V0.1.1
LoRaWAN init: ?CatenaBase::GetProvisioningStyle: failing
?CatenaWingFram2k::LoRaWAN::begin: Arduino_LoRaWAN:begin failed
failed
CPU Unique ID: f2-06-ca-96-35-50-4d-51-20-20-20-35-37-1e-11-ff
**** no platform, check provisioning ****
No mods detected
vBat:    4299 mV
EV_JOINING
EV_TXSTART
EV_JOINED
NwkID:   00000013   DevAddr: 26022f77
EV_TXSTART
~~~

3) can the USB serial data be re-assigned to a standard UART on the device, like Serial1
4) Is there a list of operatingflags avilable?
5) is there a document that explaned the steps needed to add a new board to the stack?

thanks...

terrillmoore, 2019-06-07:
Sorry, I've been busy on a project.  My colleagues will try to help with some of this; otherwise I'll try to reply over the weekend.

dhineshkumarmcci, 2019-06-10:
Hi @trlafleur, 
Please find our answers inline in this comment.
- I noticed on start up, I see LoRaWAN Init: message, where is the ?CatenaBase::GetProvisioningStyle: infor comming from.
`The log message "LoRaWAN init: ?CatenaBase::GetProvisioningStyle: failing" represents that the network parameters(appeui, deveui and appkey) are not configured. This message comes from "CatenaBase_GetProvisioningStyle.cpp" in the Catena-Arduino-Platform library.`
https://github.com/mcci-catena/Catena-Arduino-Platform/blob/553dd7f81c236699e91161d18b76a9bca5bb5a90/src/lib/CatenaBase_GetProvisioningStyle.cpp#L82-L89
- can the USB serial data be re-assigned to a standard UART on the device, like Serial1
`Unfortunately, you cannot re-assign the USB Serial to Generic Serial with the current version of Board Support Package.`
- Is there a list of operatingflags avilable?
`Yes, we have a list of Operatingflags and you can find them in "CatenaBase.h" of Catena-Arduino-Platform library.`
https://github.com/mcci-catena/Catena-Arduino-Platform/blob/553dd7f81c236699e91161d18b76a9bca5bb5a90/src/CatenaBase.h#L144-L152
- is there a document that explaned the steps needed to add a new board to the stack?
`Hardware design documents can help you in this case, you can find them from our Github repository "HW-Designs". Please refer the schematic of Catena 4460 sensor shield.`
https://github.com/mcci-catena/HW-Designs/blob/master/Boards/Catena-4460/100001204b_(Catena-4460-MCCI-LW-Feather-Sensor-Shield-Schematic).pdf

We are working on your other question on FRAM. We will get back to you soon with the answer.

Best Regards,
Dhinesh

terrillmoore, 2019-06-23:
The 2k FRAM support comes from a separate library, which we forked from adafruit: https://github.com/mcci-catena/Adafruit_FRAM_I2C.  The address is set in the begin() routine, Adafruit_FRAM_I2C::begin(); if the input address is 0, the hardware default is used.

So you have to change:
https://github.com/mcci-catena/Catena-Arduino-Platform/blob/553dd7f81c236699e91161d18b76a9bca5bb5a90/src/lib/Catena_Fram2k.cpp#L77-L79

to be:

```c++
        if (! this->Super::begin() ||
	    ! this->m_hw.begin(MY_I2C_ADDRESS, &Wire))
		return false;
```

Define MY_I2C_ADDRESS to the address you're using on the board, and this should work.

terrillmoore, 2019-06-23:
On the SAMD BSP, it is possible to use Serial1. See how I do it in the https://github.com/mcci-catena/Modbus-for-Arduino/blob/master/examples/catena_simple_host/catena_simple_host.ino.  You will have to figure out how to substitute Serial1 for Serial as needed. Once you do that, things should work; we've tested the command framework on other boards that use HW serial. There may be software compatibility issues with Serial1 vs SerialUSB, namely the `while (! Serial)` idiom. On our STM32 BSP, we added a `!` operator on the UartSerial ports (returning constant `true`). We can consider doing that for the SAMD as well, but that's a separate ticket.