mcci-catena/Catena-Arduino-Platform #253: ReadAnalog fails

drpuur, 2020-06-11:
I am using MCCI 4610 and see "?CatenaStm32L0::ReadAnalog(3): CatenaStm32L0_ReadAnalog() failed (5)" messages as in https://github.com/mcci-catena/Catena-Arduino-Platform/issues/125

This happens not all the time and seems to start happening only after some time.

Board Version: 2.7.0

Libraries:
| URL | Commit | Commit Date |
| --- | -----  | ----------- |
| https://github.com/mcci-catena/arduino-lmic.git  | 6fe04ec | Tue, 12 May 2020 09:16:47 -0400  |
| https://github.com/mcci-catena/arduino-lorawan.git  | 4bc0d48 | Sat, 9 May 2020 12:38:28 -0400  |
| https://github.com/mcci-catena/Catena-Arduino-Platform.git  | 92019ca | Tue, 12 May 2020 01:34:08 -0400  |
| https://github.com/mcci-catena/Catena-mcciadk.git  | a428006 | Sat, 21 Dec 2019 20:45:26 -0500  |
| https://github.com/mcci-catena/MCCI_FRAM_I2C.git  | f0a5ea5 | Sat, 21 Dec 2019 16:17:01 -0500  |

I read the battery voltage on every loop (after waking up from deep sleep):
```
  // vBat
  int vbat_mv = (int)(gCatena.ReadVbat() * 1000.0f);
  res.vbat = GetVBatValue(vbat_mv);
  if (config_data.debug_level > 0) {
    gCatena.SafePrintf("vBat:    %d mV\n", vbat_mv);
  }
```
Attached is the log with the relevant lines.
[ReadAnalog-errors.txt](https://github.com/mcci-catena/Catena-Arduino-Platform/files/4763260/ReadAnalog-errors.txt)

dhineshkumarmcci, 2020-06-11:
@drpuur Thank you for bringing in the issue. Right time you raised this, We are already working on it. We have pushed a patch in the branch [issue253](https://github.com/mcci-catena/Catena-Arduino-Platform/tree/issue253). Can you please pull changes to your libraries and let us know how it works?

drpuur, 2020-06-11:
@dhineshkumarmcci  Great! I have just recompiled with branch issue253 and will let you know about the results.

drpuur, 2020-06-11:
Not sure if this is related, but since this patch the node does not join any longer. I see the join requests in the console of the Lorawan Network (Swisscom), but it looks like the node never receives the join accept (even though SF12 is used).

I then did a reset of the board (using the switch), and now it looks like it lost the provisioning.

I attach the full log.
[serial-swisscom-11jun2020-1.txt](https://github.com/mcci-catena/Catena-Arduino-Platform/files/4765492/serial-swisscom-11jun2020-1.txt)

drpuur, 2020-06-11:
It's pretty strange: I did a reset again (this time with "system reset"), and now the node came up just fine and also joined the network. I will capture the output again and report back.

drpuur, 2020-06-12:
So far no errors, but I noticed a regression: no matter if I plug USB charging cable or not, it measures about the same value for vBus:

with USB power cable attached:
```vBus:    1614 mV```

no USB cable attached:
```vBus:    1619 mV```

This is where that measurement comes from:
```
  // vBus
  float vBus = gCatena.ReadVbus();
  gCatena.SafePrintf("vBus:    %d mV\n", (int)(vBus * 1000.0f));
```
Since this is used to detect if the USB cable is plugged (fUsbPower) and I use this to decide if I go in deep sleep or not, this is pretty important for my use case that this works.

dhineshkumarmcci, 2020-06-12:
Hi Joerg,
 
We would like to know the vBus value logged before these changes in library.
Can you also please make sure that no change in library is made at your end after cloning?
 
We tried to replicate the issue at your end. We have made stress test to behaviour at our end. 
We used Catena 4610, with library change in branch ‘issue253’. The result at our end is as below:
Without USB connected -              vBus:    3366 mV
With USB connected -                    vBus:    5072 mV

drpuur, 2020-06-12:
Hello Dhinesh

Sorry, must be something else. I deleted the comment after I made it, as it is not related with your patch. I have this on 2 of 4 4610 nodes, not sure what it is. I desoldered those and reused them on another PCB so it might be a hardware problem. When measuring with the multimeter I get correct values though. Maybe I just have to dump these, it‘s just a bit odd that two behave like that.

Best regards
—
Joerg

PS: I will report tomorrow if it still runs fine, until now there have been no problems.

drpuur, 2020-06-13:
The error did not happen anymore so it really looks like this has been fixed. See Log.

[ReadAnalog.log](https://github.com/mcci-catena/Catena-Arduino-Platform/files/4774602/ReadAnalog.log)

dhineshkumarmcci, 2020-06-15:
@drpuur Thank you for the update.

terrillmoore, 2020-06-26:
This still has not been addressed. The reversion was never pushed again, but the docs were pushed.

terrillmoore, 2020-06-26:
@mcciyssaroha please coordinate straightening this out.

commit 53b7e26c (2020-06-15): Fix #253: change comment to call AdcStart() before each readings
commit ec2fc01b (2020-06-15): Merge pull request #254 from mcci-catena/issue253

Fix #253: Use AdcStart() before each ReadAnalog value
commit 5b231f9a (2020-06-15): Revert "Fix #253: Use AdcStart() before each ReadAnalog value"
commit 6e9209cf (2020-06-15): Merge pull request #255 from mcci-catena/revert-254-issue253

Revert "Fix #253: Use AdcStart() before each ReadAnalog value"
closed 2020-07-31 by terrillmoore