mcci-catena/Catena-Arduino-Platform #271: Release v0.20.0

terrillmoore, 2020-12-23:
Update README, etc.  See delta in #123 for details.

- [ ] update library.properties version to 0.20.0
- [ ] update `CATENA_ARDUINO_PLATFORM_VERSION` in CatenaBase: change here:
    https://github.com/mcci-catena/Catena-Arduino-Platform/blob/1de835c93c4f76908e6fba119732fe45d58db7f6/src/CatenaBase.h#L58
- [ ] update README.md version for badges (line 7)
- [ ] update README.md release history (towards end).
- [ ] generate pull request
- [ ] merge into master.
- [ ] tag and release.

terrillmoore, 2020-12-23:
@dhineshkumarmcci, if you can do everything up through generating the PR, I'll so the merge, tag, release.

All, for the "release title", should we say "Add MCCI Catena 4802 Support"?

dhineshkumarmcci, 2020-12-23:
@terrillmoore I have made the version changes for release and submitted the merge request #272 
I think we can title as "New board 4802 and more control to Si1133".

commit b373538d (2020-12-23): Fix #271: prepare for v0.20.0 release
closed 2020-12-23 by terrillmoore