mcci-catena/Catena-Arduino-Platform #300: SAMD boards compilation gets failed with latest release

dhineshkumarmcci, 2021-06-21:
With the latest version of library, SAMD boards get build issues. Please find the error logs below.

```
In file included from C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src/Catena4450.h:46:0,
                 from C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src/Catena.h:23,
                 from C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\examples\catena_hello_lora\catena_hello_lora.ino:16:
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src/CatenaWingFram2k.h:150:22: error: 'virtual void McciCatena::CatenaWingFram2k::LoRaWAN::NetSaveSessionState(const SessionState&)' cannot be overloaded
         virtual void NetSaveSessionState(
                      ^
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src/CatenaWingFram2k.h:147:15: error: with 'virtual void McciCatena::CatenaWingFram2k::LoRaWAN::NetSaveSessionState(const SessionState&)'
  virtual void NetSaveSessionState(
               ^
```

terrillmoore, 2021-06-21:
Your commit 7fe5c72 seems to be what's needed.

terrillmoore, 2021-06-21:
When doing the pull request, please set version to 0.21.1-1 (as it will be pre-release 1 for 0.21.1).

dhineshkumarmcci, 2021-06-21:
However above fix, did not help SAMD compilation to succeed. We also get build issue on Catena Download which uses Flash in STM32.
```
In file included from C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src\lib\Catena_Download.cpp:22:0:
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src/Catena_Download.h: In static member function 'static constexpr const char* McciCatena::cDownload::getStateName(McciCatena::cDownload::State_t)':
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src/Catena_Download.h:118:9: error: body of constexpr function 'static constexpr const char* McciCatena::cDownload::getStateName(McciCatena::cDownload::State_t)' not a return-statement
         }
         ^
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src\lib\Catena_Download.cpp: In member function 'McciCatena::cDownload::State_t McciCatena::cDownload::fsmDispatch(McciCatena::cDownload::State_t, bool)':
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src\lib\Catena_Download.cpp:368:36: error: 'FLASH' was not declared in this scope
                 if (HAL_IS_BIT_SET(FLASH->PECR, FLASH_PECR_PELOCK))
                                    ^
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src\lib\Catena_Download.cpp:368:49: error: 'FLASH_PECR_PELOCK' was not declared in this scope
                 if (HAL_IS_BIT_SET(FLASH->PECR, FLASH_PECR_PELOCK))
                                                 ^
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src\lib\Catena_Download.cpp:368:66: error: 'HAL_IS_BIT_SET' was not declared in this scope
                 if (HAL_IS_BIT_SET(FLASH->PECR, FLASH_PECR_PELOCK))
                                                                  ^
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src\lib\Catena_Download.cpp:370:46: error: 'FLASH_PEKEY1' was not declared in this scope
                     WRITE_REG(FLASH->PEKEYR, FLASH_PEKEY1);
                                              ^
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src\lib\Catena_Download.cpp:370:58: error: 'WRITE_REG' was not declared in this scope
                     WRITE_REG(FLASH->PEKEYR, FLASH_PEKEY1);
                                                          ^
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src\lib\Catena_Download.cpp:371:46: error: 'FLASH_PEKEY2' was not declared in this scope
                     WRITE_REG(FLASH->PEKEYR, FLASH_PEKEY2);
                                              ^
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src\lib\Catena_Download.cpp:374:17: error: 'FLASH' was not declared in this scope
                 FLASH->PECR &= ~FLASH_PECR_ERASE_Msk;
                 ^
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src\lib\Catena_Download.cpp:374:33: error: 'FLASH_PECR_ERASE_Msk' was not declared in this scope
                 FLASH->PECR &= ~FLASH_PECR_ERASE_Msk;
                                 ^
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src\lib\Catena_Download.cpp:377:21: error: 'FLASH_FLAG_EOP' was not declared in this scope
                     FLASH_FLAG_EOP|FLASH_FLAG_WRPERR|FLASH_FLAG_PGAERR|\
                     ^
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src\lib\Catena_Download.cpp:377:36: error: 'FLASH_FLAG_WRPERR' was not declared in this scope
                     FLASH_FLAG_EOP|FLASH_FLAG_WRPERR|FLASH_FLAG_PGAERR|\
                                    ^
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src\lib\Catena_Download.cpp:377:54: error: 'FLASH_FLAG_PGAERR' was not declared in this scope
                     FLASH_FLAG_EOP|FLASH_FLAG_WRPERR|FLASH_FLAG_PGAERR|\
                                                      ^
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src\lib\Catena_Download.cpp:378:21: error: 'FLASH_FLAG_SIZERR' was not declared in this scope
                     FLASH_FLAG_SIZERR|FLASH_FLAG_OPTVERR|FLASH_FLAG_RDERR|\
                     ^
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src\lib\Catena_Download.cpp:378:39: error: 'FLASH_FLAG_OPTVERR' was not declared in this scope
                     FLASH_FLAG_SIZERR|FLASH_FLAG_OPTVERR|FLASH_FLAG_RDERR|\
                                       ^
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src\lib\Catena_Download.cpp:378:58: error: 'FLASH_FLAG_RDERR' was not declared in this scope
                     FLASH_FLAG_SIZERR|FLASH_FLAG_OPTVERR|FLASH_FLAG_RDERR|\
                                                          ^
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src\lib\Catena_Download.cpp:379:21: error: 'FLASH_FLAG_FWWERR' was not declared in this scope
                     FLASH_FLAG_FWWERR|FLASH_FLAG_NOTZEROERR
                     ^
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src\lib\Catena_Download.cpp:379:39: error: 'FLASH_FLAG_NOTZEROERR' was not declared in this scope
                     FLASH_FLAG_FWWERR|FLASH_FLAG_NOTZEROERR
                                       ^
[... cut ...]
C:\Users\dhineshkumar\Documents\Arduino\libraries\Catena-Arduino-Platform\src\lib\Catena_Download.cpp:385:36: error: 'FLASH_SR_BSY' was not declared in this scope
                 while (FLASH->SR & FLASH_SR_BSY)
                                    ^
```

Hopefully, adding #if directive for STM32 architecture (ARDUINO_STM32) for `Catena_Download.h` and `Catena_Download.cpp` would fix this issue. On using `#ifdef ARDUINO_STM32` in `Catena_Download` source files, able to fix above issue and make build successful.

@terrillmoore would like to have your advice on adding #if directive for these files.

terrillmoore, 2021-06-21:
Yes, well, that's sort of expected, given that we have not implemented the bootloader on SAMD21 at all. For now, add `#if`, but we should refactor so that the STM32 code is separated out and in the stm32 subdirectory. The downloader should be calling a hardware-specific routine to set the EEPROM flag.

dhineshkumarmcci, 2021-06-21:
Thanks Terry, have made the changes and submitted a pull-request #301

commit 69673834 (2021-06-21): Fix #300: use STM32 architecture for Catena Download
commit e9aac162 (2021-06-21): Fix #300: remove unused virtual void NetSaveSessionState()
commit 7fe5c72a (2021-06-30): Fix #300: remove unused `virtual void NetSaveSessionState()`
commit dcac8940 (2021-06-30): Merge pull request #301 from mcci-catena/issue300

Fix #300: fix SAMD build issues with latest release
closed 2021-06-30 by terrillmoore