mcci-catena/Catena-Arduino-Platform #326: Connecting Catena to Helium Network

mhmayyan, 2021-12-10:
Previously, I was able to transmit data to the helium network. Now, I am unable to receive data but the device keeps trying to join. I also tried to add a new device but it also got stuck at the joining request as shown below.

```
EV_JOINING
NwkID:   00000000   DevAddr: 00000000
EV_TXSTART
EV_JOIN_TXCOMPLETE
EV_TXSTART
EV_JOIN_TXCOMPLETE
```

It keeps repeating this forever (EV_TXSTART and EV_JOIN_TXCOMPLETE). I am able to see the join requests on the Helium console but it doesn't send the payload. I selected Helium for the LoRaWAN network and used the default for the subband. Please advise. 

```
system version 
Board: Catena 4612
Platform-Version: 0.21.2
Arduino-LoRaWAN-Version: 0.9.1
Arduino-LMIC-Version: 4.1.0
MCCIADK-Version: 0.2.2
MCCI-Arduino-BSP-Version: 3.0.4
```

Thanks.

closed 2021-12-11 by mhmayyan

terrillmoore, 2021-12-11:
Is this fixed?

mhmayyan, 2021-12-12:
Yes, I already opened the same issue under [arduino-lmic](https://github.com/mcci-catena/arduino-lmic/issues/828), which is the right place, I think.
Thank you, Terry.

mrx23dot, 2022-02-21:
Please close ticket.